Desuuuu/klipper #1: Move axes doesnt work

ffnull, 2020-09-19:
Hello. Installs your mod. All fine exept moving axes. Icon doesnt toucheble

Desuuuu, 2020-09-19:
Your configuration is missing some sections required for the screen logic to work.

The missing sections are the `[pause_resume]` section and possibly the `[probe]` section.

Please have a look at [these example configurations](https://github.com/Desuuuu/DGUS-reloaded-Klipper-config) for working configurations.

I'll implement some additional checks for proper configuration soon.

Desuuuu, 2020-09-19:
As of 4be47eb9aa8d367046b364eae39b4b4280377c98, sanity checks are performed during startup. This should prevent issues while using the screen.

Also, the `[bed_mesh]` and `[probe]` sections which were required before are now optional and the `[pause_resume]` section doesn't need to be set explicitly anymore. With these changes, your current configuration should now work as it is.

To update, please switch back to the master branch and update with these commands:

```bash
cd ~/klipper
git checkout master
git pull
sudo service klipper restart
```

closed 2020-09-19 by Desuuuu

ffnull, 2020-09-19:
Confirm. Right now works perfect. Thnx for your work! Did you plain merge branch with original klipper?

Desuuuu, 2020-09-19:
Yes, I did merge the most recent upstream changes into my branch while I was at it.

ffnull, 2020-09-19:
Another issue. What iv made wrong? Progress display starts and ends. all adjust of print - works. But doesnt show percentage.

Desuuuu, 2020-09-19:
Progress display relies solely on external data provided by `M73`.

If you're using OctoPrint, you can use [this plugin](https://plugins.octoprint.org/plugins/m73progress/) to send it automatically.

ffnull, 2020-09-19:
Great. Thnx you!