Desuuuu/klipper #2: Unknown config object 'gcode_macro'

socalswimmer, 2020-09-22:
was running KevinOConnor/klipper without any issues on my CR-10S Pro. Decided to try your firmware for the touchscreen. installed as per instructions, first the screen, then cloned your fork and flashed the printer. added touchscreen section to my existing printer.cfg, klipper wont start. unknown config object 'gcode_macro' i tried removing all gcode_macro reference from printer.cfg but that doesnt help. klippy.log attached.

[klippy.log](https://github.com/Desuuuu/klipper/files/5259394/klippy.log)

Desuuuu, 2020-09-22:
This should be solved as of 1419cc236e8b258956b6349230c3aed1849f89d8. Let me know if you encounter any other issue.

closed 2020-09-23 by Desuuuu

socalswimmer, 2020-09-23:
Sorry for a late reply, but yes, indeed this solved the problem. Thank you.