Desuuuu/klipper #6: __level_manual_point enhancement to use Klipper defined locations

ReXT3D, 2021-01-19:
First of all I wanted to send a **HUGE** thank you and congratulations for creating the best looking and most usable interface for the CR-10S Pro.  Your work, without any doubt, provides the most functional and professional looking touch interface for the printer.

I initially started with your Marlin port, but just recently decided to try Klipper and I may stick with it for good based on what I have seen so far.  I have gone through complete setup and optimization of my CR-10S Pro with both firmware flavours and I have several suggestions - this is one of them.  But please be gentle with me as I am new to Klipper and I also don't know much about Python.

The manual level code in **__level_manual_point** should follow the Klipper configuration values, if they are defined in printer.cfg.  Specifically, location 1 (data == 1) should use [safe_z_home] location and movement parameters if they have been defined, and locations 2 through 5 should use the [bed_screws] locations.  In absence of user defined values within printer.cfg, the code should default to the values as currently defined.

Desuuuu, 2021-01-29:
I don't feel like it would add that much value to follow the `bed_screws` configuration, especially considering how tedious it would be to implement properly.

What would you think about having the option to configure the front/rear/left/right distance separately for the screen in the `[t5uid1]` section instead?

ReXT3D, 2021-01-29:
I think that would certainly be a good workaround.  Without knowledge of Python and how Klipper is structured, I have no appreciation for the complexity of the implementation.

I spent quite a bit of time getting all the bed and travel dimensions correct for the CR-10S Pro and my current `bed_screws` configuration lands the nozzle exactly over each screw.

Desuuuu, 2021-01-30:
This is now available in the master branch (see a8e13181a27db78f9052fbb160b89d6b65922817). The [example configuration file](https://github.com/Desuuuu/klipper/blob/master/config/sample-t5uid1.cfg) is also up-to-date.

Thanks for the configuration, I'll have a look and update my example configurations later.

closed 2021-01-30 by Desuuuu

ReXT3D, 2021-01-30:
Awesome, thank you very much!

With respect to configurations, I also posted an "issue" in your configurations repository.  I would be happy to contribute by supplying my optimized CR-10S Pro configuration so that you can update your example.  The primary tweaks in it relate to the bed measurements (correct travel extents and stop locations for symmetrical and centred print area), bed mesh extents to symmetrically cover almost the whole bed while avoiding the clips, etc.

I also significantly reduced the z approach speed for maximum probing accuracy with a capacitive probe.  Long time ago I replaced the Creality probe with an industrial capacitive probe made by Autonics, Part Number CR18-8DN that made a huge amount of difference in first layer consistency.  But my recent adventures in self configured firmware, including Marlin, started to show some small inconsistencies in first layer height, on the order of 0.07mm.  So with Klipper I was able to easily perform a parametric study evaluating the accuracy and repeatability of z stop based on approach speed, homing retract amount and speed, heating on or off, etc.  My current settings have been absolutely bullet proof and I get perfect first layer height with the capacitive probe.

ReXT3D, 2021-01-30:
I tested the latest code and it works perfectly.  The one very minor outstanding item are the XY coordinates of the centre point.  Ideally it should be the same as `[safe_z_home]`, which in my configuration is:

```
[safe_z_home]
home_xy_position: 178,150  # this is for 300x300 print surface with 28mm probe to nozzle offset
```

Cheers.