Desuuuu/klipper #8: Definition of M73 in t5uid1.py conflicts with display_status.py preventing use of Fluidd

ReXT3D, 2021-01-29:
While exploring Klipper with your excellent display support for CR-10S Pro I decided to give Fluidd a try.  Fluidd requires inclusion of [display_status] in printer.cfg, which loads klippy/extras/display_status, which in turn defines M73 gcmd.  This conflicts with definition of M73 that is also found in klippy/extras/t5uid1/t5uid1.py, resulting in "**gcode command M73 already registered**" warning during Klipper start.  It would be awesome if both could be used at the same time.

Desuuuu, 2021-01-29:
I just pushed a change to the [display_status](https://github.com/Desuuuu/klipper/tree/display_status) branch that should allow my code to work alongside the `display_status` module.

I haven't had the time to test it just yet but you're more than welcome to try it.

ReXT3D, 2021-01-29:
That's awesome, thank you!  I am in the middle of a print this afternoon but will try to test the changes over the weekend.  Sorry for putting this in the wrong repository by the way, I had my 6.5 year old climbing all over me...

ReXT3D, 2021-01-30:
Alright.  Print finished and I swapped the Raspberry Pi, cloned the correct branch and restarted everything.  Klipper now starts fine!

Fluidd (almost) works - a few things in the web interface are broken in it, but I believe this is likely related to a different (much more recent) Klipper version from your repo vs. what Fluidd expects.  Basically Velocity, Acceleration & Accel to Decel all show `0` (zero) as values, and Extruder and Heater Bed Power both show `NaN%`.  I should probably report it in the Fluidd repository.

One thing seems broken in your new code: the printer LCD  longer shows any changes in print progress, it's stuck at 0%.  Progress does work on the Fluidd side.  Everything else seems to be working just fine and the printer is doing another print using Fluidd & your new code.  I'm even trying to push it a bit closer to the edge with ArcWelder GCODE.

Thank you again for all the hard work!

Desuuuu, 2021-01-30:
> One thing seems broken in your new code: the printer LCD no longer shows any changes in print progress, it's stuck at 0%.

Keep in mind that the host *needs to* send `M73 P` commands for the progress to update.

It does appear to work for me and the configuration conflict is no longer an issue.

ReXT3D, 2021-01-30:
Of course!  With all the back and forth I forgot that Octoprint needed a dedicated plugin for `M73 P` and Fluidd obviously does not do that by default.  Thank you for all your efforts once again!  This is awesome!

closed 2021-01-30 by ReXT3D