Desuuuu/klipper #22: I cannot 'make'

fmjensen, 2021-03-05:
When I have done `make menuconfig` and try to do a `make` I get this error:

```
pi@octopi:~/klipper $ make
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/spi_software.o
  Compiling out/src/neopixel.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/avr/main.o
  Compiling out/src/avr/timer.o
  Compiling out/src/avr/gpio.o
  Compiling out/src/avr/adc.o
  Compiling out/src/avr/spi.o
  Compiling out/src/avr/i2c.o
  Compiling out/src/avr/hard_pwm.o
  Compiling out/src/avr/watchdog.o
  Compiling out/src/avr/serial.o
  Compiling out/src/generic/serial_irq.o
  Compiling out/src/avr/serial_t5uid1.o
src/avr/serial_t5uid1.c:13:6: error: #error "The serial port selected for the T5UID1 screen is already used"
     #error "The serial port selected for the T5UID1 screen is already used"
      ^
make: *** [Makefile:64: out/src/avr/serial_t5uid1.o] Error 1
pi@octopi:~/klipper $
```

I can't figure if I am doing something wrong or if this is a bug so I hope to get help here...
My printer is a Creality Ender 5 Plus.

Desuuuu, 2021-03-05:
This is caused by a bug with the default screen serial port when advanced configuration options are hidden.

The problem should be fixed by commit bd2cad43147a58fdcdedd425a04a263f5b713cc8.

fmjensen, 2021-03-05:
Yes - that seems to have fixed it.
I executed
`git pull`
`make menuconfig`
and then `make` did it's job without any hiccups

Thank you for the swift response - now I will continue the install

closed 2021-03-05 by Desuuuu