Desuuuu/klipper #24: DWIN T5UIC1 Ender 3 V2 rotary encoder display support

ihrapsa, 2021-03-08:
Hey, Thanks for putting your time into creating this! I was wondering if you could add support to the **Ender 3 V2** dwin display. It's a color **non touch TFT** based on the **T5UIC1** mcu and it's controlled by a **rotary encoder.**
Here are some images of the display, some specs of the T5UIC1 and the datasheet of the  DMT48270C043_04WN (which is the closest I could find to the E3V2 display model: DMT48270C043_04WNZ11)

[image]

[image]

[T5UIC1_appplication_guide.pdf](https://github.com/Desuuuu/klipper/files/6104294/T5UIC1.pdf)

[DMT48270C043_04WN_datasheet.pdf](https://github.com/Desuuuu/klipper/files/6104317/DMT48270C043_04WN_datasheet.pdf)

Any guiding into modifying the existing code to fit this display would be much appreciated as well. :blush:

Desuuuu, 2021-03-08:
There isn't much similarity between T5UID1 and T5UIC1 displays. Supporting these displays would require a completely new touchscreen firmware and significant changes to the board firmware.

That's not something I plan on investing time into at the moment.

ihrapsa, 2021-03-09:
I see, that's alright. Thanks anyway!

closed 2021-03-09 by Desuuuu