Desuuuu/klipper #32: Support for ATSAMD MCU

ReXT3D, 2021-04-05:
I have been looking through your MCU code changes to try to appreciate what would be involved in porting your screen support to additional MCUs in Klipper.  On the surface it looks like it might not be that involved, but I am probably being naive due to my extremely limited coding experience.  I am tempted to tinker with it a bit, but would likely need a lot of hand holding.

I basically just threw-in the towel on the Creality control board due to its substantial limitations and rather poor design.  I have an SKR E3 Mini v2 and also a Duet 3 Mini 5+ on the way as potential replacements.  I think I will settle on the Duet since I am a bit of a freak when it comes to electronics design and quality.  Coincidentally someone just got Klipper working on the Duet 3 Mini 5+ and I would love to have the native display working with it.

Are you on the Klipper Discord channel by any chance?

Desuuuu, 2021-04-05:
It definitely isn't too complicated to add support for additional boards since they already have working serial implementations. It's just a matter of making a few modifications to get everything working together (and modifying the config menu for broader usage).

I joined the Discord, feel free to message me.

Desuuuu, 2021-04-05:
This is now available in the atsamd branch.

It compiles fine but I obviously cannot test it on a board, so please let me know if it works for you.

ReXT3D, 2021-04-06:
My goodness, you are on fire!  Thank you - I really did not expect you to do this and especially this quickly!  My Duet board will not get here until the end of this week at the earliest, and I also need to get some additional parts from DigiKey, although this is normally overnight.  As soon as I finish with the installation and have the ability to test the build I will let you know!

On a related note, is there a way to select a specific serial port?

On an unrelated note, I could not find you on Discord.  Please feel free to ping me, I am under the same handle there.

Thank you much again!

Desuuuu, 2021-04-06:
The ATSAME54P20A has 8 SERCOM interfaces (and IO multiplexing) so it's pretty flexible in terms of what you can do.

There isn't much implemented in Klipper though so it will require some changes to allow using SERCOM1-7 and different multiplexing options (currently it's only using SERCOM0 on PA10/PA11 as far as I can tell).

closed 2021-05-26 by Desuuuu

ReXT3D, 2021-05-26:
Many thanks again for working with me to get this running on the Duet 3 Mini 5+.  I can confirm that it is now working perfectly with your DGUS reloaded screen firmware.

I still think you should create a pull request to get SERCOM1-7 and different multiplexing options into mainline Klipper.

PR #34 (2021-05-26): ATSAMD support
Adds ATSAMD micro-controllers support.

Closes #32.