Desuuuu/klipper #33: Reverting to stock display firmware?

Bluetopia, 2021-04-05:
First of all, thanks for putting this together.  One of the biggest hangups I had with Klipper is the lack of machine control, and your updates helped get my touchscreen working with it. 

However, I've had to revert my recent changes to my printer (Ender 5 Plus with Silent board) as I wasn't getting the same quality as I was with stock hotend/silent board firmware and I needed to print some items. 

As part of this I wanted to revert the touch screen display changes as well.  I have tried clearing out the MicroSD card and copying the DWIN_SET files from the most recent Creality Silent board firmware release.  I get the proper main screen image, but none of the buttons are responsive, and there are no temperatures displayed.

[image]

Any thoughts on what I might need to do to revert the display update and getting it functional again?  Would including the optional T5UID1.cfg file from the wiki (https://github.com/Desuuuu/DGUS-reloaded-Klipper/wiki/Flashing-the-firmware) help here, or is that specific to your updates?  (It's binary, so I'm not certain what it actually does)

Thank you!

Desuuuu, 2021-04-05:
Flash the `T5UID1.CFG` file available [here](https://github.com/Desuuuu/DGUS-reloaded-klipper/tree/master/extras/flash_reset), then  flash your new firmware and it should work properly.

This problem happens when the `13_*.bin` file has a different name across firmwares I think.

Bluetopia, 2021-04-05:
No luck.  Tried flashing just the reset T5UID1.CFG file you linked to, then the display firmware from Creality without success.  Also tried replacing the Creality T5UID1.CFG file with the reset and flashing that (as I had done for the Klipper update originally) and no luck.  Display is still unresponsive and no temperature information is displayed.

You mention this potentially being an issue due to mismatched names.  I notice that the firmware files provided by Creality have a mismatched 13*.bin and 14*.bin file.  (Filename is in Chinese rather than English).  Would it be as simple as renaming these files to match the BMP, or are there references to the Chinese filename elsewhere that would prevent this from working?

Desuuuu, 2021-04-05:
Yes, renaming the files to `13_touch.bin` and `14_variable.bin` should solve the problem. The prefix is what really matters, except for this bug obviously.

I'm surprised it didn't work using the reset config though. You did get a message about erasing the flash on the blue flashing screen, right?

Bluetopia, 2021-04-05:
Yep, got "Formatting NAND Flash" displayed on the screen when using your cfg file.

Third time's the charm.  Renamed the Chinese files to match their English names, dropped your cfg file on the drive to ensure I'm working from a clean NAND and flashed.  Screen shows loading bar on startup, the temperatures are visible and the touch screen is responsive. 

Thanks for the help!

closed 2021-04-05 by Bluetopia