Desuuuu/klipper #36: Section 't5uid1' is not a valid config section...

jasonturcotte, 2021-04-10:
Had the display running with Octoprint front-end and decided to try Fluidd.

When I copied my config to Fluidd, I get the error

`Section 't5uid1' is not a valid config section
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted`

Once I comment that out, the printer talks to Fluidd, but I (obviously) do not have a working screen.

Desuuuu, 2021-04-11:
This error most likely means you are not running my fork.

jasonturcotte, 2021-04-11:
Forgive my ignorance, but does your fork run on the board of the printer (Ender 5) or the Raspberry Pi?

Desuuuu, 2021-04-11:
Both the microcontroller and the host firmware contain modifications.

closed 2021-04-11 by Desuuuu