Desuuuu/klipper #39: Abort/pause/resume dont work on the DGUS touchscreen

GeoCyberwolf, 2021-04-26:
Abort/pause/resume dont work on the touchscreen

Desuuuu, 2021-04-26:
These features require that your host supports the `pause`, `resume` and `cancel` [action commands](https://reprap.org/wiki/G-code#Action_commands) to work.

GeoCyberwolf, 2021-04-26:
Hi! I fixed this in my fork of this. I am trying to request to merge it in but Im not great with git :|

GeoCyberwolf, 2021-04-26:
Issue was in here: https://github.com/GeoCyberwolf/klipper/commit/970187a4f65771e1f7491d02a24f4f747325fc77

VIXIVIXIV, 2021-04-27:
I don't think this should be merged, because the original code works just fine with action commands enabled in Octo.

GeoCyberwolf, 2021-04-27:
Not everyone uses octoprint, better options are available. Most people installing klipper now use MainsailOS, its a much better lighter UI. 

 The issue isnt that actions arent enabled, its that it sends it as a message. (I can tell by the colouration of the text in the console). and when I type action:cancel etc in manually to the console it works.
 
 Removing the "RESPOND TYPE=command MSG=" before the command fixes the problem. It will likely also work with action:cancel, but I decided to use CANCEL_PRINT as this is what Klipper now expects you to use as far as I understand?

VIXIVIXIV, 2021-04-27:
I just remembered what I had to do to get it working: you have to add the action commands section to your main klipper config. You may be able to fix it just by changing the prefix for commands.
[image]

GeoCyberwolf, 2021-04-27:
Ahh, well it's probably worth putting that in the readme. Though, with my fix it should work out of the box with any klipper config.

I am going to test doing action:cancel without the prefix too. I think it will work.

GeoCyberwolf, 2021-04-27:
It doesnt work. So currently, as-is the user has to modify the Klipper config for these buttons to work.

With my changes, it uses the default klipper commands, so no modification to the Klipper config is needed.

Honestly, the whole thing should be merged into the main klipper repo, the touchscreen support is good enough now!

I'd be tempted to merge my fork because it fixes the only problem I was able to find, and is more user friendly imo. 

But I dont want to take credit for @Desuuuu's awesome work, so I will leave it up to him to decide what he wants to do.

VIXIVIXIV, 2021-04-27:
Your fix only works for the Fluiddpi implementation, or anywhere else that expects the same string for pause and cancel. In Octo for example, the string is //pause or similar. I think you should mess around with the action commands config more, though I can't replicate the response behavior in Fluid without spinning up an instance of it.

GeoCyberwolf, 2021-04-27:
So of course the problem is nothing is standardized 🙃 

I wasnt aware the frontend had anything to do with it, as CANCEL_PRIINT etc are default in Kliipper as far as I understand. 

I will do some more testing.

GeoCyberwolf, 2021-04-27:
PAUSE and RESUME are standard documented klipper gcodes: https://mmone.github.io/klipper/G-Codes.html

CANCEL_PRINT is part of the same code in Klipper, but the functionality is currently undocumented: https://github.com/KevinOConnor/klipper/issues/3895

commit 2529eea4 (2021-04-27): handle pause/resume/cancel when using virtual SD

Fixes #39
closed 2021-04-27 by Desuuuu