Desuuuu/klipper #41: Resume uses the wrong command for virtual SD prints

GeoCyberwolf, 2021-04-29:
it does action:resume instead of RESUME. 

I think this is because the sd card isnt considered active when the print is paused, as it checks printer.virtual_sdcard.is_active

I am going to test fixing this by removing that check on my fork. will report back!

Desuuuu, 2021-04-29:
This should be fixed now.

GeoCyberwolf, 2021-04-29:
> 
> 
> This should be fixed now.

pause and resume work correctly now.

Abort doesnt work if you press it when its printing, I get "SDCARD_RESET_FILE cannot be run from the sdcard" in the console. It works if you press pause first.

Desuuuu, 2021-04-29:
AFAICT, this command isn't ever called by Klipper itself. Also, this error should only occur when it's being ran by your G-code file.

GeoCyberwolf, 2021-04-29:
Well, its whats happening when I press abort during a print, if I dont pause it first. 

Workaround could be to make it pause, then cancel

Desuuuu, 2021-04-29:
The only reason this issue happens is because you've replaced the default CANCEL_PRINT command with a macro. Fix your macro and the problem will go away.

There's nothing to work around here.

closed 2021-07-07 by Desuuuu