Desuuuu/klipper #58: Enable support for moonraker update manager by adding a tag

drandarov-io, 2021-06-18:
Forks of Klipper can apparently only be properly parsed by the update manager in moonraker if a git tag is present according to a conversation with the developer.
I'm not sure if thats feasible with you setup for auto-updates, but maybe you can look into it.

drandarov-io, 2021-06-18:
[image]

Desuuuu, 2021-06-18:
I don't really tag anything on the main branch since there isn't really any "release" upstream. If that makes a difference for you, then I'll start tagging when I merge upstream changes.

Indeed, there are also the tags from the upstream sync action, but I tend to clear these after I manually merge.

drandarov-io, 2021-06-18:
It would improve compatibility with Mainsail and Moonraker, so I would be very thankful :)

Desuuuu, 2021-06-18:
Should be good now. I'll try to keep that in mind.

closed 2021-06-18 by Desuuuu