Desuuuu/klipper #62: M73 not working on Ender-6

TurboTim, 2021-07-03:
Hi,

Apologies if this is not the right place to raise this - maybe there is a Discord?

I followed the guide to install your modified Klipper and DGUS repos on my Ender-6: https://3dprintbeginner.com/how-to-install-klipper-on-creality-ender-6/

The install went well and everything is working as expected, apart from not seeing any progress on the DGUS screen, presumably since Klipper is not sending M73 updates.  I have `[display_status]` in the config, and I read somewhere this should be enough to trigger this in Klipper - am I missing something or is M73 not working in your fork as expected?

I am a dev so happy to look into a bug fix PR if you want to point me in the right direction...

Cheers
Tim

TurboTim, 2021-07-03:
Actually, found a solution soon after posting this, setting the config in PrusaSlicer to emit M73 which works great - is this something that makes sense as a Klipper native feature?

Desuuuu, 2021-07-04:
https://github.com/Desuuuu/DGUS-reloaded-Klipper/wiki/Print-progress-display

Klipper can only estimate the progress of virtual SD prints without M73. This would be fine as a fallback to M73 however I'm working on a different firmware so development here is pretty much frozen.

TurboTim, 2021-07-07:
Ooh new firmware... prey tell :-)

Setsunai, 2022-10-11:
I made a macro script that will work for any slicer and it's done through the printer.cfg and it does use the klipper virtual sdcard to print on the display and updates like every second. 
https://github.com/Desuuuu/klipper/discussions/117