Desuuuu/klipper #80: Test version not working with LPC1769 and UART0

VIXIVIXIV, 2021-10-01:
Installed clean, unmodified test branch, and latest release from accompanying DGUS_Menu firmware. Config options register fine, but shortly after "Ready" there's an e-stop and shutdown due to "Internal error during ready callback: Timeout waiting for write acknowledgement". klippy.log attached.
[klippy.log](https://github.com/Desuuuu/klipper/files/7263278/klippy.log)

Desuuuu, 2021-10-01:
Thanks for the report and the log! I was made aware of this issue a few days ago.

Your touchscreen most likely requires an update (you can find the procedure [here](https://github.com/Desuuuu/DGUSPrinterMenu/wiki/Core-firmware-update)).

You're supposed to be redirected to an update screen in this case. I'll look into the problem.

Desuuuu, 2021-10-01:
Commit e96cac17d33097036af8d80808765790249685bd should fix the problem.

You should now be redirected properly in case an update is required.

closed 2021-10-01 by Desuuuu