Desuuuu/klipper #118: Naive time estimation leads to an unhandled exception in Klippy (dgus-display)

dkabot, 2022-04-11:
Using the `dgus-display` branch, I experienced a failure very shortly into my first print (after switching branches).
The log of the crash and surrounding dumps can be found here: https://gist.github.com/dkabot/a2cab71991dff3a91445c3cc5d4a2ddf

The underlying error is a TypeError in `dgus_status.py` on line `79`: The variable `finish_at_naive` at this point contains a tuple that the code is trying to change a value in:
https://github.com/Desuuuu/klipper/blob/dgus-display/klippy/extras/dgus_status.py#L79
Tuples are immutable, so this fails and causes a swift crash of Klippy and print failure before finishing the first layer.
Nowhere else in the file does this appear to occur; everything either assigns the whole tuple as a value or simply `None`.

I've worked around this locally by simply changing the line to assign a new tuple constructed from the values of the existing one, but I am not a primarily Python developer and cannot speak as to if this is a proper solution:
```
self.finish_at_naive = (self.finish_at_naive[0], self.finish_at_naive[1] + pause_duration)
```

Desuuuu, 2022-04-11:
Good catch, thanks!

Would you like to make a PR or should I fix it?

dkabot, 2022-04-11:
Gone ahead and created a PR so this can be properly held against me in the future.
I printed a calibration cat successfully post-edit, so it should hopefully at least suffice to not be a rude awakening for others.

closed 2022-04-11 by Desuuuu