Desuuuu/klipper #127: Can't compile it for SKR E3 V3

kamil9489, 2022-09-18:
I just cant compile it for SKR Mini E3 V3 with "Enable DGUS T5UID1 screen option" enable. I'm getting a bunch of error whatever serial port i choose:

pi@fluiddpi:~/klipper $ make
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/spi_software.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/stm32/watchdog.o
  Compiling out/src/stm32/gpio.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/armcm_boot.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_reset.o
  Compiling out/src/generic/timer_irq.o
  Compiling out/src/stm32/stm32f0_timer.o
  Compiling out/src/stm32/stm32g0.o
  Compiling out/src/stm32/gpioperiph.o
  Compiling out/src/stm32/stm32f0_adc.o
  Compiling out/src/stm32/stm32f0_i2c.o
  Compiling out/src/stm32/spi.o
  Compiling out/src/stm32/stm32f0_serial.o
  Compiling out/src/generic/serial_irq.o
  Compiling out/src/stm32/t5uid1/serial.o
src/stm32/t5uid1/serial.c: In function 't5uid1_USARTx_IRQHandler':
src/stm32/t5uid1/serial.c:50:27: error: 'USART_TypeDef' {aka 'struct <anonymous>'} has no member named 'SR'; did you mean 'ISR'?
     uint32_t sr = USARTx->SR;
                           ^~
                           ISR
src/stm32/t5uid1/serial.c:51:15: error: 'USART_SR_RXNE' undeclared (first use in this function); did you mean 'USART_ISR_RXFF'?
     if (sr & (USART_SR_RXNE | USART_SR_ORE))
               ^~~~~~~~~~~~~
               USART_ISR_RXFF
src/stm32/t5uid1/serial.c:51:15: note: each undeclared identifier is reported only once for each function it appears in
src/stm32/t5uid1/serial.c:51:31: error: 'USART_SR_ORE' undeclared (first use in this function); did you mean 'USART_ISR_ORE'?
     if (sr & (USART_SR_RXNE | USART_SR_ORE))
                               ^~~~~~~~~~~~
                               USART_ISR_ORE
src/stm32/t5uid1/serial.c:52:32: error: 'USART_TypeDef' {aka 'struct <anonymous>'} has no member named 'DR'; did you mean 'RDR'?
         t5uid1_rx_byte(USARTx->DR);
                                ^~
                                RDR
src/stm32/t5uid1/serial.c:53:14: error: 'USART_SR_TXE' undeclared (first use in this function); did you mean 'USART_ISR_TXFE'?
     if (sr & USART_SR_TXE && USARTx->CR1 & USART_CR1_TXEIE) {
              ^~~~~~~~~~~~
              USART_ISR_TXFE
src/stm32/t5uid1/serial.c:53:44: error: 'USART_CR1_TXEIE' undeclared (first use in this function); did you mean 'USART_CR1_TXFEIE'?
     if (sr & USART_SR_TXE && USARTx->CR1 & USART_CR1_TXEIE) {
                                            ^~~~~~~~~~~~~~~
                                            USART_CR1_TXFEIE
src/stm32/t5uid1/serial.c:45:22: error: 'USART_CR1_RXNEIE' undeclared (first use in this function); did you mean 'USART_CR1_RXFFIE'?
                    | USART_CR1_RXNEIE)
                      ^~~~~~~~~~~~~~~~
src/stm32/t5uid1/serial.c:57:27: note: in expansion of macro 'CR1_FLAGS'
             USARTx->CR1 = CR1_FLAGS;
                           ^~~~~~~~~
src/stm32/t5uid1/serial.c:59:21: error: 'USART_TypeDef' {aka 'struct <anonymous>'} has no member named 'DR'; did you mean 'RDR'?
             USARTx->DR = data;
                     ^~
                     RDR
src/stm32/t5uid1/serial.c: In function 't5uid1_enable_tx_irq':
src/stm32/t5uid1/serial.c:45:22: error: 'USART_CR1_RXNEIE' undeclared (first use in this function); did you mean 'USART_CR1_RXFFIE'?
                    | USART_CR1_RXNEIE)
                      ^~~~~~~~~~~~~~~~
src/stm32/t5uid1/serial.c:66:19: note: in expansion of macro 'CR1_FLAGS'
     USARTx->CR1 = CR1_FLAGS | USART_CR1_TXEIE;
                   ^~~~~~~~~
src/stm32/t5uid1/serial.c:66:31: error: 'USART_CR1_TXEIE' undeclared (first use in this function); did you mean 'USART_CR1_TXFEIE'?
     USARTx->CR1 = CR1_FLAGS | USART_CR1_TXEIE;
                               ^~~~~~~~~~~~~~~
                               USART_CR1_TXFEIE
src/stm32/t5uid1/serial.c: In function 't5uid1_init':
src/stm32/t5uid1/serial.c:76:35: error: 'USART_BRR_DIV_Mantissa_Pos' undeclared (first use in this function); did you mean 'USART_CR2_DIS_NSS_Pos'?
     USARTx->BRR = (((div / 16) << USART_BRR_DIV_Mantissa_Pos)
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                   USART_CR2_DIS_NSS_Pos
src/stm32/t5uid1/serial.c:77:37: error: 'USART_BRR_DIV_Fraction_Pos' undeclared (first use in this function); did you mean 'USART_CR2_DIS_NSS_Pos'?
                    | ((div % 16) << USART_BRR_DIV_Fraction_Pos));
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                     USART_CR2_DIS_NSS_Pos
src/stm32/t5uid1/serial.c:45:22: error: 'USART_CR1_RXNEIE' undeclared (first use in this function); did you mean 'USART_CR1_RXFFIE'?
                    | USART_CR1_RXNEIE)
                      ^~~~~~~~~~~~~~~~
src/stm32/t5uid1/serial.c:78:19: note: in expansion of macro 'CR1_FLAGS'
     USARTx->CR1 = CR1_FLAGS;
                   ^~~~~~~~~
In file included from out/board-generic/board/armcm_boot.h:4,
                 from src/stm32/t5uid1/serial.c:8:
src/stm32/t5uid1/serial.c:41:23: error: 'USART3_IRQn' undeclared (first use in this function); did you mean 'USART1_IRQn'?
   #define USARTx_IRQn USART3_IRQn
                       ^~~~~~~~~~~
src/ctr.h:36:22: note: in definition of macro 'DECL_CTR_INT'
         (REQUEST), { args }, 0 }
                      ^~~~
out/board-generic/board/armcm_boot.h:10:58: note: in expansion of macro 'CTR_INT'
     DECL_CTR_INT("DECL_ARMCM_IRQ " __stringify(FUNC), 1, CTR_INT(NUM))
                                                          ^~~~~~~
out/board-generic/board/armcm_boot.h:14:9: note: in expansion of macro 'DECL_ARMCM_IRQ'
         DECL_ARMCM_IRQ(FUNC, (NUM));                    \
         ^~~~~~~~~~~~~~
src/stm32/t5uid1/serial.c:79:5: note: in expansion of macro 'armcm_enable_irq'
     armcm_enable_irq(t5uid1_USARTx_IRQHandler, USARTx_IRQn, 0);
     ^~~~~~~~~~~~~~~~
src/stm32/t5uid1/serial.c:79:48: note: in expansion of macro 'USARTx_IRQn'
     armcm_enable_irq(t5uid1_USARTx_IRQHandler, USARTx_IRQn, 0);
                                                ^~~~~~~~~~~
make: *** [Makefile:64: out/src/stm32/t5uid1/serial.o] Error 1

In comparison compile for SKR E3 V2 works without any problems.

osvatha, 2022-09-18:
I have the same issue, config:
*] Enable extra low-level configuration options
    Micro-controller Architecture (STMicroelectronics STM32)  --->
    Processor model (STM32G0B1)  --->
    Bootloader offset (8KiB bootloader)  --->
    Clock Reference (8 MHz crystal)  --->
    Communication interface (USB (on PA11/PA12))  --->

nvu2, 2022-10-08:
Was looking to install this onto my SKR mini E3 V3. Same error response when compiling the FW.