Desuuuu/klipper #158: Update to newer Klipper

ale-chronics, 2024-06-07:
Hi ! 
Thanks again for your great work on this, I use it every day with my Ender 6  and I'd love for this project to remain active !
Would you be able to merge this with a more recent version of Klipper ? 
Thanks a lot !

xdadrm, 2024-07-24:
[klipper_updater.txt](https://github.com/user-attachments/files/16364059/klipper_updater.txt)

@ale-chronics , maybe you find this attached bash script useful.
I've used that for the last two years or so to keep my klipper installation in sync with the original klipper and retaining the Ender 6 patches from Desuuuu.

Simply run the script where the ./klipper/ directory is located.

( you may want to be sure that you have a backup of that klipper folder first. )

Keep in mind that  newer klipper versions also expect new microcode on the controller :
cd klipper; make clean; make    (might require make menuconfig, to update the .config file)

nipunarora8, 2024-11-23:
@xdadrm Will this work for CR10s pro. I have desuuuu's klipper installed on my board, rpi and the screen. Can you guide me how to update it?
Would be a great help as I am thinking to install eddy probe which is only supported with mainline klipper + I can keep my klipper updated. 
Thanks

zimmertr, 2024-11-23:
I recently wrote this guide for installing upstream Klipper on a stock Ender 6. It just requires changing out the touchscreen: https://github.com/zimmertr/Ender-6-Klipper-Guide

xdadrm, 2024-11-24:
> @xdadrm Will this work for CR10s pro. I have desuuuu's klipper installed on my board, rpi and the screen. Can you guide me how to update it? Would be a great help as I am thinking to install eddy probe which is only supported with mainline klipper + I can keep my klipper updated. Thanks

@nipunarora8  As I do not have a CR10, I'll not say "it works and here's how" ...hope the following still helps :

**Klipper Software**
All the script above does it transfer desuuuu's patches into upstream klipper and I use that rather regularly to upgrade my installation. So as long you can get your original klipper directory back, there is little you can break by just trying (a recent backup is always a good idea anyway).

**Klipper Firmware on the MCU**
Now that is a little trickier, if you're already on a 0.12 version (check klippy*logs for "Last MCU build version: v0.12.0"), if
it is lower on your MCU, communication between  Klipper and the MCU will probably not work after the first step.
enter the klipper directory, copy .config from your old klipper directory if  you started fresh, and try "make clean; make".
If you get errors, run "make menuconfig" to update .config with new options, usually going with the defaults is ok.
Once the firmware compiles, use out/klipper.bin to upgrade your CR10 MCU, via SDCard.
(also here, having your original .bin file available would allow you to downgrade back to where you have been.)

**Display**
I do not recall having had to upgrade the display after the initial conversion, so that sould continue to work as is.

@zimmertr 
A nice, well put together guide !   @nipunarora8 appears to not be using a PI Display but the patched original Creality Display.

xdadrm, 2024-11-26:
@chirrupted  if you had to add the [t5uid1[ section to your config that sounds like you were NOT running desuuuu's klipper on your setup before ?!  

**If that is the case, do not continue,  revert back to your old klipper installation and look at @zimmertr 's approach. **

If you want to contine (and I don't know if the CR10s pro is  the same as the Ender 6 - MCU and Display wise ..) :

The error sounds like your MCU firmware is not up-to-date - you could check with "grep T5 klipper/.config" it should look like this :

CONFIG_T5UID1_SERIAL=y
CONFIG_STM32_T5UID1_SERIAL_USART3=y
CONFIG_STM32_T5UID1_SERIAL_USART3_PB11_PB10=y

if it is not, use make menuconfig to change the settings, make  and then reflash the mcu with the new compiled binary.

chirrupted, 2024-11-26:
@xdadrm I use an Ender 5 Plus. anyways, I got it to work (just forgot to flash to the printer :P) however I can't click on the Move button and retracting and advancing filament doesn't work. when printing, the screen doesn't change either (stays on homescreen). before using the script everything worked.

xdadrm, 2024-11-26:
@chirrupted Hah, you got me there I've got you mixed  up with @nipunarora8  :)

Anyway, glad it 'semi-worked' for you - from your description, I can only imagine that these might be incorrect  for your machine / MCU,

CONFIG_STM32_T5UID1_SERIAL_USART3=y
CONFIG_STM32_T5UID1_SERIAL_USART3_PB11_PB10=y

chirrupted, 2024-11-26:
@xdadrm thanks! though i think I'll be using an android phone instead :P sorry for wasting your time

nipunarora8, 2024-11-26:
@xdadrm Thank you for your ideas. CR10s pro has the same screen as the ender 6 (I guess). And if I am not wrong, desuuuu also has Cr10s pro (because it was his config which I took while setting up klipper). 
I will implement your ideas on my printer but I am very busy until the weekend. Will keep you updated. Thanks

nipunarora8, 2024-11-28:
Hey @xdadrm, I seem to use your script but no success. I put your script next to klipper folder, ran it, two scripts were having conflicts (src/atsamd/serial.c and src/atsamd/Kconfig) so I used the new version. The make and flashing went without errors. For flashing it generated a .elf file which I flashed using **make flash FLASH_DEVICE=serial port**

Now my klipper version in mainsail says klipper -> v0.0.0-4766-gdgus-reloaded-230526-0-gbaae7f3a-dirty-inferred

and I can't connect to mcu cuz I have an error **gcode command SET_PRINT_STATS_INFO already registered**

Do you have an idea what can I do to make it work? 
Thanks

brumela, 2024-12-06:
> @ale-chronics
> 
> @ale-chronics, you might find the attached bash script helpful. I've been using it for the past two years to keep my Klipper installation updated with the original Klipper while maintaining the Ender 6 patches from Desuuuu.
> 
> Just run the script in the directory where your ./klipper/ is located. (Make sure you have a backup of your Klipper folder first.)
> 
> Also, remember that newer versions of Klipper may require updated firmware on the controller: ( cd klipper; make clean; make (might require make menuconfig, to update the .config file )
> 
> ```
> bash script :
> 
> #!/bin/bash
> if [ ! -d klipper/.git ]; then
>   git clone https://github.com/Klipper3d/klipper.git klipper
> else
>   echo "Klipper exists .."
> fi
> 
> pushd klipper 
> 
> 
> if ( git branch | grep desuuuu-ender6 -q ); then
>   git switch desuuuu-ender6
> else
>   git branch desuuuu-ender6  
>   git checkout desuuuu-ender6
> fi
> 
> GIT_EDITOR=/bin/true git pull https://github.com/Klipper3d/klipper.git
> GIT_EDITOR=/bin/true git pull -q -X ours --no-rebase https://github.com/Desuuuu/klipper.git
> 
> popd
> ```

I did update to latest with your script, script works ok and also no errors when building... But after flashing new firmware, move (middle) button, and some buttons in setup are not working. Anyone with same problems?

My printer is Cr10sPro, with latest Desuuuu version of klipper screen was working Ok.

xdadrm, 2024-12-06:
@brumela first thing might be to try and start fresh, the script is 'dumb' and rather forcefully simply merges the two repositories together - often good enough but not failsafe. 

Try moving your klipper directory to kipper.broken, and run the script again - that will pull mainline klipper and merge desuuuu's patches in.   Then, best copy your original .config (or kipper.broken/.config) to klipper/.config , rebuild and flash the firmware to be sure to have that match the klipper software.

tssk, 2024-12-15:
@xdadrm Thank you for the script. I used it and it works great.

I am wondering if someone with the knowledge maybe could setup new repository that will be updated for the others that are not so skilled? :)

xdadrm, 2025-01-14:
> @xdadrm Thank you for the script. I used it and it works great.
> 
> I am wondering if someone with the knowledge maybe could setup new repository that will be updated for the others that are not so skilled? :)

@tssk , @Desuuuu , @brumela 

I gave this a shot and set up a fork that should sync desuuuu's patches with upstream klipper every tuesday (if there were changes). We'll see if and how that works in reality.

[klipper-Desuuuu-patch](https://github.com/xdadrm/klipper-Desuuuu-patch)

brumela, 2025-01-15:
> I gave this a shot and set up a fork that should sync desuuuu's patches with upstream klipper every tuesday (if there were changes). We'll see if and how that works in reality.
> 
> [klipper-Desuuuu-patch](https://github.com/xdadrm/klipper-Desuuuu-patch)

Thanks. So fork syncing is based on your script?
If so, I doubt that will work in my machine CR10s PRO.

Well, I also try to merge (sync) desuuuu's branch with upstream klipper, following this [guide](https://www.freecodecamp.org/news/how-to-sync-your-fork-with-the-original-git-repository/), but didn't succeed.
What I notice is that sync with [v0.12.0](https://github.com/Klipper3d/klipper/tree/v0.12.0) works, but with latest doesn't. 
So, I guess latest changes brakes desuuuu's code. I also examine merged changes, made manual merge, didn't succeed.

But I will definitely try to build from your synced repo and report back.

xdadrm, 2025-01-15:
@brumela if my script from above did not work, then this repo will unfortunately not work for you either - the code in both directories (from the script and from this repo ) would be exactly the same.

When you tried the merge yourself, you likely encountered a few conflicts that my script rather forcibly resolves and while that works in my situation, might have negative effects for your case.

I'll have a look to check if there might be anything 'obvious' ..

xdadrm, 2025-01-15:
@brumela 

I might have something - you're using the atmega2560 (and #160 seems be similar for the same mcu) , so I looked for changes in that area.

Try adding this at line 13 of src/avr/Kconfig - right after select HAVE_STRICT_TIMING :
select HAVE_GPIO_BITBANGING if !MACH_atmega168

and rebuild/flash the firmware on the mcu. (make menuconfig, make ..)

--- why ? ---
Attempting to merge desuuuu with klipper shows the following intersting changes :

CONFLICT (content): Merge conflict in src/atsamd/Kconfig
CONFLICT (content): Merge conflict in src/atsamd/serial.c
Auto-merging src/avr/Kconfig

For the atmega2560, the two atsamd files should not be relevant, but src/avr/Kconfig is. Looking at changes there we see that HAVE_GPIO_BITBANGING has been removed :

>     select HAVE_GPIO_BITBANGING if !MACH_atmega168
14d14
<     select HAVE_LIMITED_CODE_SIZE if MACH_atmega168 || MACH_atmega328 || MACH_atmega328p || MACH_atmega32u4
150a151,152
> 
> source "src/avr/t5uid1/Kconfig"

xdadrm, 2025-01-15:
@brumela 
If all else fails, the following might help pinpoint the  commit (or at least date range) where the problem started with, with that information we might be able to fix the problem. Tedious.

```
# Start with Desuuuu klipper
git clone https://github.com/Desuuuu/klipper.git klipper

pushd klipper
git remote add upstream https://github.com/Klipper3d/klipper.git
git fetch upstream
# git checkout origin/dgus-reloaded

# This first ones hould not change anything ( https://github.com/Klipper3d/klipper/commits/master/ - Aug 25, 2022 - the last klipper commit that made it into Desuuuu's repo. )
git merge -X ours f7e29b276e5c2454eaa801e78cf3f6aff29c4ba9 -m "starting point"
git diff HEAD..origin/HEAD > ../diff_start.txt
### make clean, make menuconfig, make, flash .. and test

git merge -X ours v0.11.0 -m "v0.11.0"
git diff HEAD..origin/HEAD > ../diff_start-v0.11.0.txt
### make clean, make menuconfig, make, flash .. and test

git merge -X ours v0.12.0 -m "v0.12.0"
git diff HEAD..origin/HEAD > ../diff_start-v0.12.0.txt
### make clean, make menuconfig, make, flash .. and test

git merge -X ours upstream/master -m "master"
git diff HEAD..origin/HEAD > ../diff_start-master.txt
### make clean, make menuconfig, make, flash .. and test
```

Once we know at shich release it stops working, we can use this to get back to that starting point and then the git merge -X ours <commit-hash> as the usual make ... to try out commits between the releases to get closer to the culprit :

```
git reset --hard f7e29b276e5c2454eaa801e78cf3f6aff29c4ba9
git merge -X [commit-hash] -m "my message"
### make clean, make menuconfig, make, flash .. and test
```

brumela, 2025-01-15:
> I gave this a shot and set up a fork that should sync desuuuu's patches with upstream klipper every tuesday (if there were changes). We'll see if and how that works in reality.
> 
> [klipper-Desuuuu-patch](https://github.com/xdadrm/klipper-Desuuuu-patch)

As expected code from your repo didn't work.

brumela, 2025-01-15:
> @brumela
> 
> I might have something - you're using the atmega2560 (and #160 seems be similar for the same mcu) , so I looked for changes in that area.
> 
> ( after re-checking; I might be wrong. after running make menuconfig , .config always contains CONFIG_WANT_GPIO_BITBANGING=y )
> 
> Try adding this at line 13 of src/avr/Kconfig - right after select HAVE_STRICT_TIMING
> 
> ```
> select HAVE_STRICT_TIMING
> select HAVE_GPIO_BITBANGING if !MACH_atmega168
> ```
> 
> and rebuild/flash the firmware on the mcu. (make menuconfig, make clean, make ..)
> 
> --- why ? --- Attempting to merge desuuuu with klipper shows the following intersting changes :
> 
> ```
> CONFLICT (content): Merge conflict in src/atsamd/Kconfig
> CONFLICT (content): Merge conflict in src/atsamd/serial.c
> Auto-merging src/avr/Kconfig
> ```
> 
> For the atmega2560, the two atsamd files should not be relevant, but src/avr/Kconfig is. Looking at changes there we see that HAVE_GPIO_BITBANGING has been removed :
> 
> ```
> >     select HAVE_GPIO_BITBANGING if !MACH_atmega168
> 14d14
> <     select HAVE_LIMITED_CODE_SIZE if MACH_atmega168 || MACH_atmega328 || MACH_atmega328p || MACH_atmega32u4
> 150a151,152
> > 
> > source "src/avr/t5uid1/Kconfig"
> ```
Also didn't work.

brumela, 2025-01-15:
I see this error when Move button is pressed,
_printer_data/logs/klippy.log_ 

`Error evaluating 't5uid1_routine __move:script': AttributeError: 'BLTouchProbe' object has no attribute 'multi_probe_pending'
Traceback (most recent call last):                                                                                                                                                                                                                                                        File "/home/brumela/klipper-Desuuuu-patch/klippy/extras/gcode_macro.py", line 61, in render
    return str(self.template.render(context))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/brumela/klippy-env/lib/python3.11/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()                                                                                                                                                                                                                                                   File "/home/brumela/klippy-env/lib/python3.11/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "/home/brumela/klippy-env/lib/python3.11/site-packages/jinja2/_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "<template>", line 2, in <module>
  File "/home/brumela/klipper-Desuuuu-patch/klippy/extras/t5uid1/t5uid1.py", line 866, in is_busy
    return (self.probe is not None and self.probe.multi_probe_pending)
                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                                                                                                                   AttributeError: 'BLTouchProbe' object has no attribute 'multi_probe_pending'
Error evaluating 't5uid1_var __switch_page:script': gcode.CommandError: Error evaluating 't5uid1_routine __move:script': AttributeError: 'BLTouchProbe' object has no attribute 'multi_probe_pending'`

After some searching I found this file klippy\extras\probe.py was problematic, so I also try to fix it but no success... 
If I find some time will try this [2592833930](https://github.com/Desuuuu/klipper/issues/158#issuecomment-2592833930) as you suggested and yes Tedious...
Anyway thanks for suggestions.

xdadrm, 2025-01-15:
Ah that's good - that looks ike it's only in the python code, not the mcu firmware , so easier to fiddle with :)

As a very first guess - try using klippy/extras/bltouch.py from upstream klipper ..

xdadrm, 2025-01-20:
@brumela 
I've made a change to [xdadrm/klipper-Desuuuu-patch)](https://github.com/xdadrm/klipper-Desuuuu-patch) in t5uid1/t5uid1.py

that should fix your problem.
Line 866 :   return (self.probe is not None and self.probe.multi != "OFF")

[Commit 237488f](https://github.com/xdadrm/klipper-Desuuuu-patch/commit/237488fa67c4689b5dd27c46ea45fa6309babd1d)

brumela, 2025-01-23:
> [@brumela](https://github.com/brumela) I've made a change to [xdadrm/klipper-Desuuuu-patch)](https://github.com/xdadrm/klipper-Desuuuu-patch) in t5uid1/t5uid1.py
>
> that should fix your problem. Line [...]

Thanks @xdadrm, Move button and also Settings > Leveling button are working after this fix.

brumela, 2025-01-23:
@xdadrm On more question, is "Screen status" working during printing with this fork?
It worked for me till [v0.12.0](https://github.com/Klipper3d/klipper/tree/v0.12.0).

alegnn, 2025-01-24:
> > [@xdadrm](https://github.com/xdadrm) On more question, is "Screen status" working during printing with this fork? It worked for me till [v0.12.0](https://github.com/Klipper3d/klipper/tree/v0.12.0).
> 
> I think it does, but haven't checked lately - will try and check when I print something next time,
> 
> and .. thanks for the feedback above, much appreciated !

It is working ! Just flashed it on my E6 and everything is working fine, thanks a lot !

brumela, 2025-01-24:
> It is working ! Just flashed it on my E6 and everything is working fine, thanks a lot !

I followed this https://github.com/Desuuuu/DGUS-reloaded-Klipper/wiki/Print-status but doesn't work.
Can you share your configs. Where did you put DGUS_PRINT_START, DGUS_PRINT_END, in slicer or in klipper printer.cfg?

I also noticed that the Pause and Cancel buttons do not work while printing.

alegnn, 2025-01-24:
In the slicer I have just `START PRINT` and `END PRINT`
The two commands `DGUS_PRINT_START` and `DGUS_PRINT_END` are in the `START_PRINT `and `END_PRINT ` macros inside the Klipper configuration.
I have modified my config a lot so it is not the best example to copy but I can answer your questions.

xdadrm, 2025-01-24:
@alegnn @brumela 

Thanks for the feedback, great to hear !
While I haven't recently confirmed that this still works, Like @alegnn I have these in printer.conf and START_PRINT/END_PRINT in the Slicer :

[gcode_macro CANCEL_PRINT]
rename_existing: BASE_CANCEL_PRINT
gcode:
    {% if 't5uid1' in printer %}
    DGUS_PRINT_END
    {% endif %}

[gcode_macro START_PRINT]
    {% if 't5uid1' in printer %}
    DGUS_PRINT_START
    {% endif %}
    ....

[gcode_macro END_PRINT]
    {% if 't5uid1' in printer %}
    DGUS_PRINT_START
    {% endif %}
    ....

tssk, 2025-02-07:
> [@tssk](https://github.com/tssk) , [@Desuuuu](https://github.com/Desuuuu) , [@brumela](https://github.com/brumela)
>
> I gave this a shot and set up a fork that should sync desuuuu's patches with [...]

@xdadrm I just cloned the repo you prepared, rebuilt and flashed klipper to my Ender6 and evrything works great! Even the Move button on the screen started to work (it stopped some time back but I do not really use it so did not mind).

Thanks again for your work!

MichielioZ, 2025-04-22:
@xdadrm Thanks for creating that repo, but unfortunately I have a small issue ...
I added your repo as KIAUH custom repo on my Ender 7 and selected it and it seems like everything worked except bed mesh. When I select "probe" in the auto leveling menu, Klipper crashes and shuts down.

Before this, I was running, I was running with [this repo](https://github.com/gbkwiatt/klipper), which worked fine for a long time, but like I mentioned in [this issue](https://github.com/fryc88/klipper-sv06plus-screen/issues/15) it stopped working with the movement and leveling menus, so I reverted and stopped updating ... (also, that repo seems to be stuck at v0.12?)
Yours works fine with the movement menu and the leveling menu opens normally, it seems to just be the actual auto-probing that crashes Klipper (I haven't tested everything else yet, just went straight to test what failed before...)

(For anyone interested with an Ender 7: I flashed the stock screen with this [Sermoon D1](https://github.com/Desuuuu/DGUS-reloaded-Klipper/discussions/33#discussioncomment-2933877) firmware. You have to rename the 000_screens.icl to 32_BaseScreen.icl and use the Desuuuu flash_reset before flashing to make it load all the updated files.)

xdadrm, 2025-04-22:
@MichielioZ if you can check for and share any error messages that might be in klipper logs when klipper 'crashes and shuts down' , I can try and have a (quick) look. Perhaps it is as simple a change as the other one .. no promises though.

MichielioZ, 2025-04-22:
@xdadrm  I checked the logs myself and I'm pretty sure I found the culprit ... In `t5uid1.py` it tries to read `self.bed_mesh.bmc.probe_helper.results`, but apparently `probe_helper` doesn't exist anymore.
The `bmc` is a reference to the BedMeshCalibrate class in the `bed_mesh.py` and it looks like perhaps the `probe_helper` has been replaced by `probe_mgr` (=ProbeManager class). Now if only I could find what the `results` should look like (which I have trouble finding), I might be able to find the equivalent in the manager class.
Hope you'll be able to take a look too, as I'm not sure if I can figure it out myself ...

MichielioZ, 2025-04-23:
Unfortunately that didn't work for me, I get the following error:

```
Error evaluating 't5uid1_var level_probing_icons:script': AttributeError: 'BedMeshCalibrate' object has no attribute 'get_mesh'
Traceback (most recent call last):
  File "/home/dietpi/klipper/klippy/extras/gcode_macro.py", line 67, in render
    return str(self.template.render(context))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/klippy-env/lib/python3.11/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/home/dietpi/klippy-env/lib/python3.11/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "/home/dietpi/klippy-env/lib/python3.11/site-packages/jinja2/_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "<template>", line 1, in <module>
  File "/home/dietpi/klipper/klippy/extras/t5uid1/t5uid1.py", line 769, in probed_matrix
    matrix = self.bed_mesh.get_mesh().get_probed_matrix()
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'BedMeshCalibrate' object has no attribute 'get_mesh'
```

I think it's weird that it's referring to `BedMeshCalibrate` when there's no clear reference to `bmc` or the class itself.
I also tried using `self.bed_mesh.bedmesh.get_mesh().get_probed_matrix()`, since `bedmesh` is a mesh reference, but that gave me the same error.

(By the way, I renamed your variable `probed_matrix` with just `matrix` as it was inside the definition of `probed_matrix` itself)

xdadrm, 2025-04-23:
Ah yes thanks - in those case you'd return 0  - this should take care of that, you  might have to play with that a bit :)

def probed_matrix(self):
    if self.bed_mesh is None:
        return 0
    mesh = self.bed_mesh.get_mesh()  # Get mesh object first
    if mesh is None:  # Check if mesh exists
        return 0
        
    # For manual probing
    if hasattr(self.bed_mesh.bmc.probe_helper, 'manual_results'):
        count = len(self.bed_mesh.bmc.probe_helper.manual_results)

    # For automatic probing
    else:
        probed_data = mesh.get_probed_matrix()  # Now safe to call
        count = sum(1 for row in probed_data for point in row if point is not None)

.. and then the rest

xdadrm, 2025-04-23:
@MichielioZ well done & thanks for the feedback ! 
Please feel free to submit a PR over in my repo if you think it's good to go - I would appreciate that ! :)

MichielioZ, 2025-04-23:
@xdadrm I also got it working with some simplified code:
```
count = len(self.bed_mesh.status['probed_matrix'][0]);
```
Again, no check marks on the graphics of the probe grid, but automatic probing seems to work fine with this one line of code too. Could you check if I read the code correctly and this one line can be used instead ?

xdadrm, 2025-04-23:
@MichielioZ , I understand that this would fail if the mesh would happen to be either not rectangular ot would contain points that are not probed , so the version with sum() should be more robust.

Unfortunately I cannot immediately identify the place where these tickmarks would be set :(

MichielioZ, 2025-04-23:
In the UI on the screen when probing, you see a square grid (5x5) representing points to probe.
Previously whenever a point was probed, a "tick" icon (="check" icon) would appear at the point on the grid that just got probed.
So either the previous `count = len(self.bed_mesh.bmc.probe_helper.results)` was different to the current `count`, or there is a setting that I'm not finding in the code to update the UI once a point is probed ...

(Maybe helpful: I also tried `count = len(self.bed_mesh.bmc.probe_mgr.base_points)`, which resulted in probing still working, but with the full grid filled with all "tick" icons ...)

MichielioZ, 2025-04-23:
I know I'm being difficult here, but I like it when code is as simple as possible :-)
```
probed data = self.bed_mesh.status['probed_matrix']
count = sum(1 for row in probed_data for point in row if point is not None)
```
As far as I can tell, that should be the same matrix and you lose an if statement and 2 function calls, or did I miss something ?

As for the user interface when probing, I found the scripts that run in the `dgus_reloaded` folder in the file `vars_out.cfg`.
On line 219 it creates the grid using `printer.bed_mesh.probed_matrix` and on line 244 I think it uses the return value of `probed_matrix` (res) to put an icon on the corresponding point on the grid.
Seems like I need to find out how that `printer.bed_mesh.probed_matrix` relates to the `probed_matrix` we were using before ...

xdadrm, 2025-04-23:
Looks elegant and should handle empty meshes, too .. :)

xdadrm, 2025-06-15:
@Desterly I've cherry-picked your change into my repo, thanks !

e043d2871e297d0952f73cb1c351de23cdf600a1

Lukasgust, 2025-07-18:
I need some help to get mt screen working properly. I've manage to flash new firmware to cr10s-pro v2,4 card (atmega2560) enable dgus. mainsail working aswell, but my screen starts and shows temperature | move | settings. with temp icons underneath but no readback. nothing happens when trying to use the screen, not even a sound. anyone that can point me in a direction. first time with klipper for me

Lukasgust, 2025-07-18:
yeap, its the klipper logo when im not connected to the printer https://imgur.com/a/by43uiN

Lukasgust, 2025-07-18:
Hope you care to read.... on top is the printer.cfg as you can see. last part of the config is for t5uid1

Starting Klippy...
Args: ['/home/user/klipper/klippy/klippy.py', '/home/user/printer_data/config/printer.cfg', '-l', '/home/user/printer_data/logs/klippy.log', '-a', '/home/user/printer_data/comms/klippy.sock']
Git version: 'v0.13.0-255-ge94a73a3'
Branch: master
Remote: origin
Tracked URL: https://github.com/xdadrm/klipper-Desuuuu-patch.git
CPU: 4 core ?
Python: '3.11.2 (main, Apr 28 2025, 14:11:48) [GCC 12.2.0]'
Start printer at Fri Jul 18 20:33:55 2025 (1752863636.0 7.8)
===== Config file =====
[stepper_x]
step_pin = PF0
dir_pin = PF1
enable_pin = !PD7
rotation_distance = 40
microsteps = 16
full_steps_per_rotation = 200
endstop_pin = ^PE5
position_endstop = 0
position_min = 0
position_max = 300
homing_speed = 50
homing_retract_dist = 5

[stepper_y]
step_pin = PF6
dir_pin = !PF7
enable_pin = !PF2
rotation_distance = 40
microsteps = 16
full_steps_per_rotation = 200
endstop_pin = ^PJ1
position_endstop = 0
position_min = 0
position_max = 310
homing_speed = 50
homing_retract_dist = 5

[stepper_z]
step_pin = PL3
dir_pin = !PL1
enable_pin = !PK0
rotation_distance = 8
microsteps = 16
full_steps_per_rotation = 200
endstop_pin = probe:z_virtual_endstop
position_min = -1
position_max = 351

[extruder]
step_pin = PA4
dir_pin = PA6
enable_pin = !PA2
rotation_distance = 22.900
microsteps = 16
full_steps_per_rotation = 200
nozzle_diameter = 0.400
filament_diameter = 1.750
max_extrude_only_distance = 500.0
max_extrude_only_velocity = 70.0
max_extrude_only_accel = 1000.0
heater_pin = PB4
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PK5
control = pid
pid_kp = 22.2
pid_ki = 1.08
pid_kd = 114
[... 115 lines cut ...]
  12  | (163.5, 157.0)  | (136.5, 155.0)
  13  | (229.2, 157.0)  | (202.2, 155.0)
  14  | (295.0, 157.0)  | (268.0, 155.0)
  15  | (295.0, 232.0)  | (268.0, 230.0)
  16  | (229.2, 232.0)  | (202.2, 230.0)
  17  | (163.5, 232.0)  | (136.5, 230.0)
  18  | (97.8, 232.0)   | (70.8, 230.0)
  19  | (32.0, 232.0)   | (5.0, 230.0)
  20  | (32.0, 307.0)   | (5.0, 305.0)
  21  | (97.8, 307.0)   | (70.8, 305.0)
  22  | (163.5, 307.0)  | (136.5, 305.0)
  23  | (229.2, 307.0)  | (202.2, 305.0)
  24  | (295.0, 307.0)  | (268.0, 305.0)
Starting heater checks for extruder
Stats 11.9: gcodein=0  mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=1300 bytes_read=5062 bytes_retransmit=9 bytes_invalid=6 send_seq=141 receive_seq=141 retransmit_seq=2 srtt=0.014 rttvar=0.002 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=15999641 heater_bed: target=0 temp=23.5 pwm=0.000 sysload=1.25 cputime=1.259 memavail=3652164 print_time=3.366 buffer_time=0.215 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000
Stats 12.9: gcodein=0  mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=1373 bytes_read=5218 bytes_retransmit=9 bytes_invalid=6 send_seq=147 receive_seq=147 retransmit_seq=2 srtt=0.012 rttvar=0.003 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=15999726 heater_bed: target=0 temp=23.6 pwm=0.000 sysload=1.25 cputime=1.263 memavail=3657840 print_time=3.366 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.1 pwm=0.000
Stats 13.9: gcodein=0  mcu: mcu_awake=0.035 mcu_task_avg=0.000154 mcu_task_stddev=0.000316 bytes_write=1379 bytes_read=5326 bytes_retransmit=9 bytes_invalid=6 send_seq=148 receive_seq=148 retransmit_seq=2 srtt=0.011 rttvar=0.005 rto=0.030 ready_bytes=0 upcoming_bytes=0 freq=15999807 heater_bed: target=0 temp=23.5 pwm=0.000 sysload=1.25 cputime=1.266 memavail=3646728 print_time=3.366 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.4 pwm=0.000
Stats 14.9: gcodein=0  mcu: mcu_awake=0.035 mcu_task_avg=0.000154 mcu_task_stddev=0.000316 bytes_write=1385 bytes_read=5432 bytes_retransmit=9 bytes_invalid=6 send_seq=149 receive_seq=149 retransmit_seq=2 srtt=0.011 rttvar=0.004 rto=0.027 ready_bytes=0 upcoming_bytes=0 freq=15999834 heater_bed: target=0 temp=23.5 pwm=0.000 sysload=1.25 cputime=1.269 memavail=3623692 print_time=3.366 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.3 pwm=0.000
Stats 15.9: gcodein=0  mcu: mcu_awake=0.035 mcu_task_avg=0.000154 mcu_task_stddev=0.000316 bytes_write=1463 bytes_read=5563 bytes_retransmit=9 bytes_invalid=6 send_seq=155 receive_seq=155 retransmit_seq=2 srtt=0.011 rttvar=0.005 rto=0.030 ready_bytes=0 upcoming_bytes=0 freq=15999931 heater_bed: target=0 temp=23.5 pwm=0.000 sysload=1.31 cputime=1.274 memavail=3615624 print_time=3.366 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.3 pwm=0.000
webhooks client 547743774736: New connection
webhooks client 547743774736: Client info {'program': 'Moonraker', 'version': 'v0.9.3-95-gb1b515a'}
webhooks: registering remote method 'shutdown_machine' for connection id: 547743774736
webhooks: registering remote method 'reboot_machine' for connection id: 547743774736
webhooks: registering remote method 'pause_job_queue' for connection id: 547743774736
webhooks: registering remote method 'start_job_queue' for connection id: 547743774736

Lukasgust, 2025-07-20:
I dont really know whats wrong, i get boot sound and the menu comes up. but screen is frozen, although pressing emergencystop updates the screen to shutdown and plays sound. so I somewhat working.

xdadrm, 2025-07-20:
Hm.

That gives me the impression it might be that the display gets data from the mcu, but the mcu does possibly not receive anything back from the display ? 
Checking connectors on mcu and display side as well as  UART config (make menuconfig), looks like UART2 is selected and correct).  Perhaps build and  flash the MCU firmware once again ?

klipper/.config should contain these lines :
CONFIG_T5UID1_SERIAL=y
CONFIG_AVR_T5UID1_SERIAL_UART2=y
CONFIG_T5UID1_SERIAL_PORT=2

Lukasgust, 2025-07-21:
Dude, I've been so frustrated that I've reimaged Raspberry Pi OS twice and flashed my printer's firmware three times. I probably spent 30hours to get things working xD. Although, thanks for the help.

ramonimbao, 2025-11-12:
>```
>    if self.probe is None:
>        return False
>    probe_owner = getattr(self.probe, "mcu_probe", self.probe)
>    multi_state = getattr(probe_owner, "multi", "OFF")
>    return multi_state != "OFF"
>```
@xdadrm Thank you! Still works for me.

ramonimbao, 2025-11-12:
Yes, I'm using Klicky PCB.

ramonimbao, 2025-11-12:
It's configured with `probe`: https://www.klipper3d.org/Config_Reference.html#probe